OLIMEX/DIY-LAPTOP #1: Would like to contribute

ZeekHuge, 2017-02-02:
Hi, I would like to contribute (as well as learn along).
Can you please elaborate the "TO DO" list at 
 DIY-LAPTOP/SOFTWARE/A64-TERES/README.md

Thanks.

felipesanches, 2017-03-09:
@ZeekHuge what are your skills? How do you see yourself contributing? Is there any specific area of the circuit/pcb design (or the computer in general) that you might be interested in improving? Perhaps (as you mentioned an interest in a "SOFTWARE" README) you may be more interested in the driver part of this project, right?

ZeekHuge, 2017-03-10:
@felipesanches Yes. I am more interested in drivers and the kernel part. Though I would be really happy to help in the circuit part too (along with learning about it).
I am an undergraduate student with Electronics and Telecommunication as my major.
I have worked on Embedded systems and was a GSoC-2016 student under BeagleBoard.org. I worked on the project called [BeagleScope](https://github.com/ZeekHuge/BeagleScope) . It involved developing support for PRUs and develop IIO drivers.

So, I would like to know if there are software gaps, bugs or requirements that I could help in ?

felipesanches, 2017-03-10:
@ZeekHuge Cool! I participated in GSoC 3 times (once as a mentor) but never on a hardware-related project.

I did work for 5 months on a project called MOD Duo (https://moddevices.com/) which used an A20-SOM. I'm not sure if the final product still uses it or if they changed the CPU, since I do not work there anymore and the motherboard design (even though it was designed with KiCAD and hosted on a private github repo) was never made public, unfortunately.

During the work at the above-mentioned startup, I did patch a portion of the ALSA sound drivers on the linux-sunxi fork of the kernel. I think perhaps https://github.com/linux-sunxi is what is used for this diy-laptop as well? Not sure. I'm still learning about it myself.

I must let you know that I'm not a mantainer of this laptop project (nor affiliated to OLIMEX in any manner), but only an enthusiast as yourself.

happy hacking,
Felipe Sanches

jonassmedegaard, 2017-11-06:
A starting point for improving Linux support for the A64 SoC used in TERES-I is http://linux-sunxi.org/Category:Development

macandchief, 2018-03-31:
I'd like to mention one thing, which is - hopefully - not an issue. I'm not sure where to write about it, so I just make a short note here.

You have the LCD cable near the hinges. This made me think of the MSI s271 notebooks, which had a "built in flaw". It had the cable also near the hinges. Every time the lid was opened or closed, the cable moved back and forth. On EVERY notebook there were broken cables after a while (the cable contained several thinner cables and sometimes the one for the speakers broke, sometimes the one for the LCD).

Im just mentioning it, because it's good to think about these design flaws and avoid them. Hopefully it is not an issue on the Teres I.

jcstaudt, 2018-05-18:
I'm not at all the arbiter for this, but the most commonly-accepted practice is:

* `CONTRIBUTING` 
  * This document should be located at the root directory of the [DIY-LAPTOP](https://github.com/OLIMEX/DIY-LAPTOP) repository and describe the best-practices for everything to be contributed to the repo
  * As this repo contains both **HARDWARE** and **SOFTWARE** codebases, the `CONTRIBUTING` guide ought to link to other more-specific documentation located within each subdirectory
  * Regardless of experience, abiding by these guidelines should give a developer the proper knowledge to generate and commit reputable contributions
* `ISSUES`
  * @macandchief Things like this are highly important design considerations.  I've personally owned 5+ devices that self-destructed this way - remember the RAZR flip phones?
* `ISSUE TAGS`
  * A great tool for project managers to use is a leverage a `good first issue` tag on issues placed that are either trivial or require a limited scope of familiarity with the nature of the project
* `README`
  * @ZeekHuge Expressing the direction and intent of a project at the highest levels is greatly appreciated.  It takes a decent chunk of time and planning to establish such project-management-level material, but external contributors may take this information and run with it - often contributing content that is within scope and merit to the overall project.

TsvetanUsunov, 2018-05-18:
@jcstaudt I add you as collaborator to this repository

jcstaudt, 2018-05-31:
I constructed a CONTRIBUTING.md document at the root directory of this repository.

I will need some help from Olimex, particularly with the ``doc/web/dev_roadmap.md`` and ``doc/web/dev_wishlist.md`` files.  I'm new here, so it's not my call to say what is coming down the line or what is most needed.

@TsvetanUsunov et al, can you please review this CONTRIBUTING guide and let me know what you think?  Can we spend a little time developing a wishlist/roadmap?

closed 2024-06-07 by TsvetanUsunov