OLIMEX/DIY-LAPTOP #2: Where are the files for the case design?

whmountains, 2017-02-03:
This is an awesome laptop!!!

I can't find the CAD models for the case.  Did I just miss them or are they closed source? 😱 Or maybe you just used an off-the-shelf model.

Thanks

DanKoloff, 2017-02-14:
We didn't design the plastic enclosure. We used a ready case. Made the PCB design after we have acquired the case in order to fit.

jcstaudt, 2018-06-16:
Closing the loop:

A 3D-printed case design has been developed by [Fork Sand, Inc.](https://github.com/forksand) and the design files may be found [here](https://github.com/forksand/olimex-teres-case).  I have added a link within this repo's [FAQ](https://github.com/OLIMEX/DIY-LAPTOP/blob/master/doc/web/res_faq.md) page.

DanKoloff, 2019-01-28:
I can no longer find the files. Seems like Fork Sand, Inc. either deleted or set the repo private. Re-opening.

DanKoloff, 2019-01-28:
I guess they migrated to other platform and now the files are located here: https://code.forksand.com/forksand/olimex-teres-case

closed 2019-01-31 by DanKoloff