OLIMEX/DIY-LAPTOP #3: 3d models for the PCB components

felipesanches, 2017-03-01:
KICAD has a very nice 3d rendering feature.

I opened the PCB1 on the 3d viewer and only the bare PCB and a single component (an 0805 part with L1 reference) got rendered in the scene.

I don't think this is very important. But certainly it would be neat to have the 3D models sorted out. Maybe it is just a matter of setting it up correctly as most 3d models are already provided in libraries just like parts symbols and footprints.

TsvetanUsunov, 2017-03-01:
we have no experience with 3D, all our kiCAD libraries are on github https://github.com/OLIMEX/KiCAD
anyone who can help is welcome

felipesanches, 2017-03-01:
If contributors are welcome, then it would be wiser to keep the issue open. There's no reason to feel ashamed of an open issue. :-) Please consider reopening this, in the hope of geting more visibility for potential contributors.

jcstaudt, 2018-05-18:
If anyone can help locate some resources for me, I can probably become more useful.  I'm starting to wrap my head around [OpenSCAD](https://www.openscad.org).

felipesanches, 2018-05-18:
openscad is absolutely great! I love it!

DanKoloff, 2018-05-21:
KiCad has 3D view, it would probably be easier and more accessible if each component gets updated with height via KiCad. Our components and libraries are located here: https://github.com/OLIMEX/KiCAD (also there is again a funny answer to the re-occurring question "Why there are no 3D drawings for the components?") in the description :)

jcstaudt, 2018-05-31:
Please pardon the potentially ignorant question but, how are the [Olimex KiKad resources](https://github.com/OLIMEX/KiCAD) different than the [vanilla KiCad](https://kicad.github.io/footprints) ones or the resources recently [released by Digikey](https://github.com/digikey/digikey-kicad-library)?  Are any of them now redundant that can be re-referenced to point to upstream variants?

bkmgit, 2019-12-30:
The KiCAD 3D viewer seems to work ok. An exported image is attached. Exporting an STP file seems less reliable in KiCAD 5.1.5-1
[image]

bkmgit, 2019-12-30:
Possible solutions are 
https://hackaday.com/2015/11/08/kicad-script-hack-for-better-mechanical-cad-export/#more-177033
https://github.com/marmni/FreeCAD-PCB

closed 2020-07-14 by TsvetanUsunov