OLIMEX/DIY-LAPTOP #4: RTL8723BS requires non-free firmware

felipesanches, 2017-03-01:
Please consider replacing the wifi chipset by something that does not require proprietary firmware.

One good candidate for that could be the chipset used on the tpe-n150usb dongle: https://www.thinkpenguin.com/gnu-linux/penguin-wireless-n-usb-adapter-gnu-linux-tpe-n150usb

This chipset also requires firmware, but the corresponding source code was released as free software and is available at:
https://github.com/qca/open-ath9k-htc-firmware

More info on the Atheros AR7010/AR9280/AR9287 chipset can be found here:
https://wikidevi.com/wiki/Atheros_AR7010

TsvetanUsunov, 2017-03-01:
we have Artheros WiFi https://www.olimex.com/Products/USB-Modules/MOD-WIFI-AR9271-ANT/ for these who want to have completely blob free WiFi
unfortunately this module is quite old and big we can't put it inside laptop space

felipesanches, 2017-03-01:
Is it only an issue with the component sizes / available PCB area? Or are there other known issues? Maybe cost of the parts? Or parts availability in stock?

Even though there are USB dongles such as the one you pointed out and the TPE-N150USB that I mentioned earlier, I can see how adopting a freedom-respecting chipset on the motherboard itself could be very positive for this project, given it's being advertised as a fully-free hardware design. Full software freedom would be a *very nice* badge to showcase.

It would be good to investigate the issue a bit deeper to make sure this is really something impossible to acchieve with the current laptop plastic case.

felipesanches, 2017-03-09:
@TsvetanUsunov Could you (or perhaps some of your co-workers) provide further insight into this issue?

jcstaudt, 2018-05-31:
If we may continue this discussion and determine the best path forward, I will dedicate some time to assist its integration.

In addition, I created a "[RISC-V Laptop Working Group](https://www.loomio.org/g/THbo1zmX)" that I believe Olimex would be a fantastic fit for.

closed 2020-07-14 by TsvetanUsunov

felipesanches, 2020-07-14:
why was this closed?