OLIMEX/DIY-LAPTOP #6: Build problems

khumarahn, 2017-10-28:
Hi, I tried to follow the linux building instructions in https://github.com/OLIMEX/DIY-LAPTOP/blob/master/SOFTWARE/A64-TERES/scripts/readme.md

I am not an expert at all. I have a few comments/questions:
* I guess that "Getting source code" section needs to be updated with the new location. Why is arm trusted firmware not mirrored in olimex repo?
* There is a requirement for `gcc-4.7-arm-linux-gnueabihf` toolchain, seems that it is only used in `u_boot-new`. Can somebody comment? Why armhf when everything else is aarch64? Why gcc-4.7? It does not seem to be needed upstream.
* `./build_uboot.sh` fails with
```./build_uboot.sh: line 29: ../sunxi-pack-tools/bin/update_boot0: No such file or directory```
I updated sunxi-pack-tools with the newer https://github.com/longsleep/sunxi-pack-tools, and it worked.

khumarahn, 2017-10-28:
Also,
* `./make_simpleimage.sh teres.img 1000 ./linux-a64-3.10.104-1.tar.xz` fails with
```
Creating teres image teres.img of size 1000 MiB ...   
Using Kernel from ./linux-a64-3.10.104-1.tar.xz ...    
20+0 records in
20+0 records out
20971520 bytes (21 MB, 20 MiB) copied, 0.0116583 s, 1.8 GB/s
48+0 records in                           
48+0 records out                           
49152 bytes (49 kB, 48 KiB) copied, 0.00029765 s, 165 MB/s
960+0 records in                                           
960+0 records out                                      
983040 bytes (983 kB, 960 KiB) copied, 0.002399 s, 410 MB/s 
50+0 records in                                   
50+0 records out                                                  
52428800 bytes (52 MB, 50 MiB) copied, 0.0262658 s, 2.0 GB/s      
mkfs.fat 4.0 (2016-05-06)                                 
Total number of sectors (102400) not a multiple of sectors per track (63)!
Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
```

Here I am not sure what's going on. Is it safe to add mtools_skip_check=1 to .mtoolsrc?

hehopmajieh, 2017-10-29:
Hi Alexey, since we need 32 bit SPL/boot0 to get u-boot running/64 bit
compiler exeeds 32k size of boot0/, we must use 32 bit compiler.
Did you clone proper sunxi-pack-tools branch /this branch is forked from
longsleep, as working/.

Best regards,
Dimitar

khumarahn, 2017-10-29:
ok, thanks for the explanation about the 32 bit compiler. I hope that the requirement for the old gcc-4.7 can be lifted at some point.

I didn't quite understand you about sunxi-pack-tools. I couldn't make work the version from here https://github.com/OLIMEX/DIY-LAPTOP/tree/master/SOFTWARE/A64-TERES/sunxi-pack-tools 
(error is in the first post). Maybe it needs to be updated with a newer one from longsleep's repo?

hehopmajieh, 2017-10-29:
Hi Alexey,
i agree with you, i also finding some typos in this instructions. I will do
my best to fix it tomorrow.
Thank you.

khumarahn, 2017-11-05:
Something else missing in the instructions:
```
# cat etc/udev/hwdb.d/teres_kbd.hwdb 
evdev:input:b0003v15BAp003C*
  KEYBOARD_KEY_7006f=brightnessdown
  KEYBOARD_KEY_70070=brightnessup
```

hehopmajieh, 2017-11-06:
Readme is up to date, and i think everything should work as expected.

closed 2017-11-06 by hehopmajieh