OLIMEX/DIY-LAPTOP #7: Wifi stops working with ERROR sd_recv_rxfifo: alloc recvbuf FAIL!

khumarahn, 2017-11-04:
Occasionally wifi crashes, and the system log has hundreds of megabytes of
```
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
Nov  4 14:43:05 teres kernel: RTL871X: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!
```

I found a reference here: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1720580
and here: https://lkml.org/lkml/2017/6/12/504

Does this affect anybody else? It happened to me with the stock ubuntu, and also with my build of gentoo.

khumarahn, 2017-11-04:
To me, it happens regularly but not often. Maybe, once every 8 hours or so. It once helped to do
```
rmmod 8723bs
modprobe 8723bs
```

hehopmajieh, 2017-11-06:
Hi Alexey, did you test patch posted on lkml ?

khumarahn, 2017-11-06:
No, because on launchpad they say it did not help. Though I can still try :-)

khumarahn, 2017-11-07:
I tested the patch, it does not help indeed.

closed 2024-06-07 by TsvetanUsunov