OLIMEX/DIY-LAPTOP #9: battery charge indication

khumarahn, 2017-11-05:
I can't make the battery indication work with lxde or lxqt. It seems that it is because some standard files like `charge_now` and `charge_full` or `energy_now` and `energy_full` are missing in the device folder:
```
# ls -al /sys/class/power_supply/battery
total 0
drwxr-xr-x 3 root root    0 Nov  5 14:37 .
drwxr-xr-x 5 root root    0 Nov  5 14:37 ..
-r--r--r-- 1 root root 4096 Nov  5 14:38 capacity
-r--r--r-- 1 root root 4096 Nov  5 14:48 current_now
lrwxrwxrwx 1 root root    0 Nov  5 14:48 device -> ../../../axp81x-supplyer.47
-r--r--r-- 1 root root 4096 Nov  5 14:48 energy_full_design
-r--r--r-- 1 root root 4096 Nov  5 14:48 health
-r--r--r-- 1 root root 4096 Nov  5 14:38 model_name
-r--r--r-- 1 root root 4096 Nov  5 14:48 online
drwxr-xr-x 2 root root    0 Nov  5 14:48 power
-r--r--r-- 1 root root 4096 Nov  5 14:48 present
-r--r--r-- 1 root root 4096 Nov  5 14:38 status
lrwxrwxrwx 1 root root    0 Nov  5 14:38 subsystem -> ../../../../../../class/power_supply
-r--r--r-- 1 root root 4096 Nov  5 14:48 technology
-r--r--r-- 1 root root 4096 Nov  5 14:48 temp
-r--r--r-- 1 root root 4096 Nov  5 14:38 type
-rw-r--r-- 1 root root 4096 Nov  5 14:37 uevent
-r--r--r-- 1 root root 4096 Nov  5 14:48 voltage_max_design
-r--r--r-- 1 root root 4096 Nov  5 14:48 voltage_min_design
-r--r--r-- 1 root root 4096 Nov  5 14:48 voltage_now
```
There is instead `voltage_now`, `voltage_min_design` and `voltage_max_design`, and probably the battery indication can be patched to use them. Or maybe one can make the `energy` or `charge` files appear?

khumarahn, 2017-11-05:
One of the values is like 1000 times less than it should be:
```
$ cat /sys/class/power_supply/battery/voltage_min_design 
3300
$ cat /sys/class/power_supply/battery/voltage_max_design 
4200000
$ cat /sys/class/power_supply/battery/voltage_now
3973000
```

khumarahn, 2017-11-06:
a dirty patch for lxpanel-0.9.3: https://pastebin.com/YniYSytz. It shows the battery percentage, but does not estimate the remaining time. Partly because the battery capacity is reported as 7mWh, while I'd expect it to be 30Wh:
```
$ cat /sys/class/power_supply/battery/energy_full_design
7000
```

I see that the reported voltage changes a lot suddenly on adapter plug in/out, by like 0.2V, this is strange.

binutzu, 2018-02-10:
Just flashed image version 1.2 and while setting/tidying up the desktop again i noticed that the Mate battery charge monitoring plugin also does not recognize the  battery - says running on AC. At the same time the "default" power manager (icon) seems to work ok: on  battery + percentage.

On a different matter, there should be another way to update the system with the officially supported distro beside wiping the content of emmc...

DanKoloff, 2018-02-12:
@binutzu We are working on a better way of update.

macandchief, 2018-03-31:
May I ask: How many hours do you get approximately with the battery e.g. on 1/3 surfing, 1/3 video and 1/3 text editing?

khumarahn, 2018-03-31:
I'd say around 9 hours. Maybe shorter or longer.

binutzu, 2018-07-23:
@DanKoloff I updated my system to Ubuntu Bionic as described in posts of Olimex Forum for Teres-Laptop. It worked fine but the only thing remaining is the battery level display.
You mentioned that Olimex provides a custom charge indicator - is that a Mate Desktop specific one and is the source code available? I did not find it in this repo.

jcstaudt, 2018-12-28:
These are the only battery charge indicators I see, and believe them to be irrelevant. I have only ever seen ``battery_charge.bmp`` when the laptop is plugged in while powered down. I am using Olimex's OEM image v1.4:

[image]

The tooltip states ``System is running on AC power. No battery present`` when plugged in and when not plugged in. (This statement has often confused me on many laptops as the power input is post-DC-conversion, but that's quite out-of-scope of this issue :smiley:)

closed 2024-06-07 by TsvetanUsunov