OLIMEX/DIY-LAPTOP #10: teres1-ledctrl consumes 100% cpu

khumarahn, 2017-11-08:
I have another issue to submit.

Occasionally, `teres1-ledctrl` starts to consume 100% of the cpu (one core), and the leds do not work. O observed it on both the original ubuntu, and on my gentoo, where I compiled it myself.

This happens rarely, I have no idea what triggers it.

A screenshot:
[image]

khumarahn, 2017-11-08:
upd: turning the keyboard on and off as in #8 does trigger this. However, I do not remember how it happened with the stock ubuntu - I wasn't experimenting with anything this low level.

d3v1c3nv11, 2017-11-13:
This script fixed the issue:
```bash
#!/bin/sh
set -e

if [ "$2" = "suspend" ] || [ "$2" = "hybrid-sleep" ]; then
    case "$1" in
        pre) systemctl stop teres1-ledctrl ;;
        post) systemctl start teres1-ledctrl ;;
    esac
fi
```
create it with
```bash
sudo nano /lib/systemd/system-sleep/teres-services
```

hehopmajieh, 2017-11-13:
I think this is a workaround , not fix.  @d3v1c3nv11  take a look  
[here](https://github.com/OLIMEX/DIY-LAPTOP/blob/master/SOFTWARE/A64-TERES/teres1-ledctrl/teres1-ledctrl.c#L82)
This will die in horrible pain, if file descriptor is missing. Simple file check will do the trick.

khumarahn, 2017-11-13:
Yeap, it is not the right fix. I have a script (in #8) which unbinds the keyboard device when the lid is closed. There I also had to add a restart of the led control service.

Also, I do not use systemd. I would happily use it, but it requires a recent kernel. Our 3.10 is just too old, while older versions of systemd require to many patches to compile. Luckily for me, gentoo comes with openrc. It is a very good init system, but it does not manage suspend.

khumarahn, 2017-11-22:
#13

d3v1c3nv11, 2017-11-22:
Merged.

khumarahn, 2017-11-22:
ok, thank you. It seems, this is done.

closed 2017-11-22 by khumarahn