OLIMEX/DIY-LAPTOP #11: keyboard and touchpad not working after firmware update

binutzu, 2017-11-13:
Hello,
I was following the issues logged here and was interested in the kbd/touchpad related ones. I applied the firmware update procedure with the latest firmware.tar.gz from github. 
After executing the update command and pressing Fn+Tux+Esc, the update program completed but my keyboard and touchpad don't work anymore since then.
I tried reboot, shutdown and boot, plug the charging cable but no improvement.

#Using the debug console on audio jack i tried to upload an older firmware but the program  #fails with the message "Unable to find rebootor". It seems to me that it tries to open some USB device and does not find it. lsusb shows the following:
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 003: ID 1908:2311 GEMBIRD 
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

What does lsusb show on a working TERES-I ?
Is there some way to repair the kbd and touchpad on my device?
Thanks!

khumarahn, 2017-11-13:
Hi, the update worked fine for me, and this is my lsusb. 
```
# lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 005: ID 15ba:003c Olimex Ltd. 
Bus 001 Device 003: ID 1908:2311 GEMBIRD 
Bus 001 Device 009: ID 05e3:0727 Genesys Logic, Inc. microSD Reader/Writer
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
`15ba:003c` is the keyboard+touchpad

binutzu, 2017-11-14:
Right. I am missing that device.
Removing the battery does not help. Actually it seems that the laptop will not boot properly only with the power plug and with battery disconnected.

I checked the cables to&from the KBD PCB and look ok.
Interestingly when i disconect the cable leading to the main board no event is shown in dmesg. But when i inserted it back there was something:

[  441.966575] usb 1-1.4: new full-speed USB device number 12 using sunxi-ehci
[  442.080086] hid-generic 0003:15BA:003B.0007: hiddev0: USB HID v1.11 Device [HID 15ba:003b] on usb-sunxi-ehci-1.4/input0
[  444.601118] usb 1-1.4: USB disconnect, device number 12

For some reason the KBD controller disconnects immediately. What would cause this behaviour? There is no chance to do an update.

I would appreaciate any hints on how can i reprogram it or further troubleshoot.

DanKoloff, 2017-11-14:
@binutzu Are all ribbon cables properly connected as shown in the manual? Before and after inserting the ribbon cable the lock latches should be closed to keep the cable in place.

d3v1c3nv11, 2017-11-14:
```bash
hid-generic 0003:15BA:003B.0007: hiddev0: USB HID v1.11 Device [HID 15ba:003b] on usb-sunxi-ehci-1.4/input0
```
This is the bootloader and is active just for 2 seconds after power up. Try to make upgrade procedure using external usb keyboard. but instead of pressing Fn+Tux+esc keys you must restart PCB5 - disconnect from main board and insert it back.

binutzu, 2017-11-14:
Thanks for the feedback. Here a photo with the upper part of the cabling around the KBD PCB.
[image]

Now I use TERES with an external USB keyboard. I will try the command line options of  hid_bootloader_cli.c

d3v1c3nv11, 2017-11-14:
To start bootloader you can reset board shorting C5.

binutzu, 2017-11-14:
Great! i have again keyboard and touchpad!

Thanks all for support. I made a change to hid_bootloader_cli so that it does not try to hard reset the device by default. This way it waits for the bootloader and has time (2s?) to write the firmware.

hid_bootloader_cli should have an option to avoid the hard-reboot and just wait

closed 2017-11-14 by binutzu

khumarahn, 2017-11-14:
Off topic: @binutzu teach me how to use the audio jack console? What is the simplest setup, what to buy, what is the pinout on the audio jack?

binutzu, 2017-11-14:
@khumarahn Actually for me it was easier than i tought. I had a USB Serial TTL adapter with 3 wires and an old audio jack with 3 contacts. Using a multimeter I found what was the ground wire on each cable  and connected those. Then i was left with two other wires which means 2 combinations. I was lucky and got it right from the beginning. Otherwise just try the other combination.
You will get the bootloader messages by default and if debug=on in uEnv.txt then also the kernel console is available where you have tty access.

I found this very useful: [https://forum.pine64.org/showthread.php?tid=5026](url)

Using this serial console i could see that xenial, arch and even slackware miniroot is booting on the laptop and is a matter of loading automatically the modules + some config to have it run nicely