OLIMEX/DIY-LAPTOP #18: Different keyboard and mouse pin?

petterreinholdtsen, 2018-01-19:
I love the idea behind the laptop, and am really contemplating to get one.  But I would very much like to have a keyboard with a mouse pin, not a touchpad, similar to the keyboards on Thinkpads. A good keyboard with slight curves on each key (for the finger to feel if it is at the center or the edges of the keys), and a physical effect on the finger when a key press is registered is vital to be able to type very quickly.  The current keyboard is very flat, as far as I could tell from testing a friends version.  What is needed to use a different keyboard with a mouse pin, and replace the mousepad with mouse buttons?  Are there alternative keyboard suppliers available?

petterreinholdtsen, 2018-01-19:
Right.  I hope something come out of it.  I've been using Thinkpad X keyboards for the last 15-20 years, and my typing speed with such keyboard is above 300 presses per minute.  Moving to a keyboard with less finger feedback lead to a significant loss of typing speed.  As the keyboard is the main link between my head and my computer, it is quite important to me.  I prefer a mouse pin over a touchpad as the touchpad is located where I rest my hands, causing spurious mouse movement and clicks, and a mouse pin is quicker to operate when I move between windows. :)

Do you have links to the people working 3D printed plastics?  I would be perfectly willing to buy a loose thinkpad keyboard (for example a replacement part for a thinkpad X42/X230/whatnot), and use it instead of the default keyboard, but I lack the skill set required to make such modified enclosure myself.

jcstaudt, 2018-10-02:
A 3D-printed case design has been developed by [Fork Sand, Inc.](https://github.com/forksand) and the design files may be found [here](https://github.com/forksand/olimex-teres-case). I have added a link within this repo's [FAQ page](https://github.com/OLIMEX/DIY-LAPTOP/blob/master/doc/web/res_faq.md).

The OEM injection-molded plastic enclosure makes it too costly to modify in such a way that allows for configurable input devices, so a modification of the 3D-printed lower assembly would be required. There is a learning curve involved with tools such as [FreeCAD](https://forum.freecadweb.org/viewforum.php?f=36), but with some dedication and support from the community, it shouldn't be too laborious to carve out a place for a loose ThinkPad keyboard.

closed 2018-10-02 by jcstaudt