OLIMEX/DIY-LAPTOP #19: 3D files for the plastic case

abdberkana, 2018-01-23:
Hi all, can some one point me to the link where to download the 3D files of the enclosure ?

DanKoloff, 2018-01-23:
We don't have such. Refer to this blog post about the plastics: https://olimex.wordpress.com/2018/01/19/teres-i-diy-open-source-hardware-modular-hackers-laptop-update/

Quote from it: "As you all know we suck at mechanical engineering and we had to find all plastic parts from Chinese laptop supplier. Some people asked about possible 3D designs of the laptop case but we are not capable to do this..."

closed 2018-01-23 by DanKoloff

jcstaudt, 2018-06-16:
Closing the loop:

A 3D-printed case design has been developed by [Fork Sand, Inc.](https://github.com/forksand) and the design files may be found [here](https://github.com/forksand/olimex-teres-case).  I have added a link within this repo's [FAQ](https://github.com/OLIMEX/DIY-LAPTOP/blob/master/doc/web/res_faq.md) page.