OLIMEX/DIY-LAPTOP #25: other distro, besides ubuntu

albjeremias, 2018-05-23:
Hi! is the laptop supporting other distros? maybe debian or archlinux? If I want to try something like that, what do I need to keep in mind to don't brick the hardware?

DanKoloff, 2018-05-24:
There is an unofficial Gentoo image, check this forum thread: https://www.olimex.com/forum/index.php?topic=5966.0

If things go bad just download the official image to a micro SD card and boot from there.

jcstaudt, 2018-05-31:
FYI: The link I see from reading the thread is [here](https://github.com/khumarahn/teres1-gentoo).

jcstaudt, 2018-05-31:
I'll see what I can do to document supported OSes with TERES.  It's just slightly difficult for me to be super thorough because I don't actually own a TERES -- I just want to see it succeed.

albjeremias, 2018-05-31:
theres this:
 - https://wiki.debian.org/InstallingDebianOn/Olimex/Teres-I
 - https://archlinuxarm.org/about/downloads (ARMv8 AArch64 Multi-platform) might work...
 - https://github.com/khumarahn/teres1-gentoo

I downloaded the gentoo image.. burn it to a sd card.. and nothing happens... I dunno how-to install ! 
(I have experience installing linux in peepz computers on more than 100 probably...) 

 - the olimex bios is weird.. is there a button to select where to boot from?

DanKoloff, 2018-05-31:
The Gentoo author has published instructions:

"
To install to MMC, replace the contents of your /dev/mmcblk0p1 and /dev/mmcblk0p2 with what is in the respective partitions in the image. Then edit /etc/fstab and /boot/uEnv.txt.
"

> the olimex bios is weird..

TERES-I is not a conventional x86 laptop. Its has an ARM chip inside and ARM architecture, not x86. The ARM architecture is more common for tablets and smartphones (and some netbooks like Chromebook); the x86 is more common for personal computers and bigger laptops. Because of the ARM architecture, TERES-I has no BIOS either (in the traditional sense of "BIOS").

The typical way of installing the operating system of TERES-I is via a micro SD card (there is a slot on the TERES-I). It generally goes like this:

1. Download the latest image provided by Olimex
2. Download it to a micro SD card (8GB or bigger)
3. Insert the card in the laptop and start the laptop, it would boot from the micro SD card
4. A command line would appear and you execute an install script, that would move the files from the card to the eMMC memory inside
5. Remove the card, so TERES-I can now boot from the eMMC memory

> is there a button to select where to boot from?

No. There is a pre-defined priority list of boot mediums (the board would attempt to boot from different places in specific order - if it can't boot from the top priority place it would try from the second, then the third, and so on).

khumarahn, 2018-05-31:
@albjeremias hi, there is no more detailed guide, sorry. This is on purpose: I did not mean to be too user friendly, because maintaining linux for teres still requires expertise.

If you are adventurous, here are approximate (untested) instructions. Run them only if you understand what is going on and that this may kill your cat. Assuming you boot from gentoo micro sd card, run
```
mkdir /mnt/mmc-boot
mkdir /mnt/mmc-root
mount /dev/mmcblk0p1 /mnt/mmc-boot
mount /dev/mmcblk0p2 /mnt/mmc-root

mkdir /mnt/sd-root
mount --bind / /mnt/sd-root

rsync -aAxXhH --delete /boot/ /mnt/mmc-boot
rsync -aAxXhH --delete /mnt/sd-root/ /mnt/mmc-root
```
Then edit `/mnt/mmc-boot/uEnv.txt` and `/mnt/mmc-root/etc/fstab` to point to `mmcblk0` instead of `mmcblk1`.

See also [install_emmc.sh](https://github.com/OLIMEX/DIY-LAPTOP/blob/master/SOFTWARE/A64-TERES/scripts/platform-scripts/install_emmc.sh) which is a part of the official installation. It does the same and more.

TsvetanUsunov, 2018-05-31:
@albjeremias please do not take it offenseve but the questions you ask show that you have no much experience with linux, in this case may be good idea to stay with ubuntu as it's most user friendly distro for beginners.
Gentoo require you to compile everything from sources and this may be a bit confusing for novices. Arch also has not so many packages available as Ubuntu, for instance you should compile KiCad from sources in Arch as if I remember correctly the KiCad team work on Ubuntu and their nightly builds are for Ubuntu.

albjeremias, 2018-05-31:
lol! :-1: dude, i use linux since 15yrs... since fedora... i used gentoo, debian, rasppi, etc.. etc... i have much experience... im just no familiar with this ARM stuffs... and I want gentoo because i bet that is way faster than stupid ubuntu, and i hate that ubuntu sold our soulds to amazon once.

jcstaudt, 2018-05-31:
If I can bring this thread back on topic please...

I don't understand the mentality of intentional limited documentation.  It is completely valid to not provide half-complete walkthroughs which contains so many gaps that the user to becomes confused and wastes tons of time, potentially losing data.  Bold warnings are generally acceptable if there is inherent risk in a step of the process.  However, unless the products are flying off the shelves faster than they can be restocked, there is certainly value in making the product more powerful, easier to use, and more reliable.  I have never experienced thoughtfully-written documentation as being a negative factor in my professional career.

If any of you find value in this, I am happy to provide a general guide to load OSes on eMMC, as well as a comprehensive example.  If this is the case, I would appreciate some direction as to not waste time using a useless example.

More frankly, what is the completion criteria of this issue?

albjeremias, 2018-06-28:
but i cant just boot some distro from the sd card?

DanKoloff, 2018-06-28:
There is an Armbian image too now. Armbian is friendlier for people outside of electronics. Check here: https://www.armbian.com/olimex-teres-a64/

sameer, 2018-10-17:
Has anyone tried to get archlinux running?

jcstaudt, 2018-12-28:
@sameer Not that I'm aware, but I'll keep an eye out until then.. and, of course, please report back if you hear of anything. It looks like there are Arch options for Olimex's sister products, but I'm sure this is not news to you: https://archlinuxarm.org/platforms

jcstaudt, 2018-12-28:
@albjeremias You definitely can. I finally got a Teres so I can be of more help now. Once you download/extract/copy a valid ``.img`` file containing a valid boot partition onto a uSD card, the Teres will look to boot off of the uSD card before eMMC. In this way, you can have a stack of imaged uSD cards handy (I coincidentally do right now!) and freely boot whatever you want. If you so desire to copy an image to the eMMC, you can do so using your own method or one of the provided scripts. This is one of the particularly pleasant aspects of ARM boards. :smiley:

These are the available Teres image torrents I currently seed. Am I missing any (apart from the laundry list of past Armbian releases)? I will create a wiki page describing the official/unofficial distros with associated torrent files. Will that be enough to satisfy this issue for now?

- Android (Android_teres_rev4.torrent)
- Armbian's latest Debian stretch xfce desktop
- Armbian's latest Ubuntu bionic xfce desktop
- @khumarahn's Gentoo lxde desktop: 20180926
- Olimex's Ubuntu (xenial/mate) releases: 20161117, 20171211, 20181220

jcstaudt, 2018-12-31:
Based on the decision from issue #32, I uploaded all known (by me personally) torrent files for the A64-TERES. I also created a README within SOFTWARE/A64-TERES describing the currently available operating system images, both Olimex-supported and third-party. Issues related to the Olimex-supported Teres images should be created within this repository, and any issues with third-party images should be submitted through their respective maintainers. Once an alternative OS is deemed production-ready for the Teres, I will be happy to link it (with disclaimers, obviously) accordingly. I think these commits satisfy the requirements for this issue, so I'm closing it. Feel free to open it again, with comment, if you disagree.

closed 2018-12-31 by jcstaudt