OLIMEX/DIY-LAPTOP #27: kernel PREEMPT, are there alternatives?

albjeremias, 2018-06-28:
**Is your feature request related to a problem? Please describe.**
I feel that my teres is quite slow, using firefox... Which kind of optimizations can I do? Can i upgrade the RAM? (i dont think so, just confirming)

**Describe the solution you'd like**
I'd like to known the command to install another kernel which is not preempt, will it affect anything? is teres using grub?

i expect something like: $ sudo apt install kernel-...-non-preempt

**Describe alternatives you've considered**
Try to install it, but i might break the bootloader? Or even the cpu may not be compatible...

q2dg, 2018-08-16:
Thanks for your clear explanation, @jcstaudt . Unfortunately, having only 2GB of RAM (and not being able to upgrade it) is a definitive black hole. Thanks anyway for your effort.

jakeogh, 2019-03-14:
The RK3328 datasheet mentions it's a 32bit memory bus on page 7: http://opensource.rock-chips.com/images/d/d7/Rockchip_RK3328_Datasheet_V1.1-20170309.pdf

Kreyren, 2023-03-02:
@q2dg Why is it a black hole? I run full GNOME and do engineering in FreeCAD on my teres without an issue O_o

closed 2024-06-07 by TsvetanUsunov