OLIMEX/DIY-LAPTOP #30: Dual display

khumarahn, 2018-12-06:
Hi, I am trying to make the HDMI work. Can't understand some things.

* What is the difference between the two .dtb files for hdmi and without hdmi?

* What does
```
		[[ -e /boot/a64/a64-olinuxino-hdmi.dtb ]] && \
		cp -f /boot/a64/a64-olinuxino-hdmi.dtb /boot/a64/a64-olinuxino.dtb
```
is `change-display` do? Which exactly files should be in `/boot/a64`?

hehopmajieh, 2018-12-07:
>    - to load modules at boot, it is easier to use a dedicated config
>    file, e.g. /etc/modules-load.d/teres-hdmi.conf; this is easier and
>    cleaner than editing /etc/modules

Hi Alexey,
please take a look at rel3 branch, this issues are fixed in this branch.
Please let me know what is the issue with   /opt/teres/hdmi/a64-olinuxino.dtb,
can
you paste boot log.
~Best, Dimitar

khumarahn, 2018-12-07:
Thanks! Is there a place to download new blobs for uboot and boot0?

hehopmajieh, 2018-12-07:
Hi Alexei,
there is no change in boot0, you can build u-boot from rel3 branch

~Dimitar

khumarahn, 2018-12-07:
ok, thanks. Unlike most of teres software, u-boot is annoying to build because the build script is ubuntu specific, and it relies on the ancient gcc-4.7. I don't have ubuntu installed. But I will install it soon.

khumarahn, 2018-12-07:
looking at
https://github.com/OLIMEX/DIY-LAPTOP/blob/672fff77303e2a79f900c536b460872f7a7f8d06/SOFTWARE/A64-TERES/scripts/install_kernel.sh#L89
the new `uEnv.txt` is very different from before. There is no `root=` setting any more, which is used in initrd.

khumarahn, 2018-12-08:
I can't make the laptop boot from an sd card with the new uboot and uEnv.txt and other files in /boot. I see a dark screen with nothing. Boot log: https://pastebin.com/TJwbBEkN

khumarahn, 2018-12-08:
Cleaner boot log: https://pastebin.com/M87YXY2z

I built the "simple image" following instructions in the rel3 branch, and compared the first 20 MiB to my image. They are only different in the first 512 bytes, the MBR, which is expected. The simple image also does not boot from the sd card.

khumarahn, 2018-12-08:
What is the logic for determining the root partition now? Do the scripts in initrd still make sense?

khumarahn, 2018-12-08:
I managed to build a working u-boot. Not sure what was wrong the first time.

I suppose I should have hdmi working now, but I don't have a mini hdmi adapter to test.

To add to all my questions above, what could be the reason hdmi does not work with xrandr?

hehopmajieh, 2018-12-10:
Hi,
dual display uses Allwinner disp2 module, this module is not drm module so
it does not support randr extensions.
Now I am trying to make drm drivers to work , but this will take some time.

~Best

khumarahn, 2018-12-13:
I received a mini-hdmi adapter. But can't make hdmi work. System log shows:
```
Dec 13 21:42:36 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 128
Dec 13 21:42:36 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 127
Dec 13 21:42:36 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 126
Dec 13 21:42:37 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 125
Dec 13 21:42:37 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 124
Dec 13 21:42:37 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 123
...
Dec 13 21:42:38 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 3
Dec 13 21:42:38 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 2
Dec 13 21:42:38 teres kernel: [HDMI WRN] file:drivers/video/sunxi/disp2/hdmi/hdmi_bsp_sun50iw1p1.c,line:600:    ddc read timeout, byte cnt = 1
Dec 13 21:42:38 teres kernel: HDMI set ptbl: 16 2 0 96 7 5 4 1 4 0 0 128 24 88 44 56 45 1 1 
Dec 13 21:42:38 teres kernel: hdmi_clk_enable_prepare()L181
Dec 13 21:42:38 teres kernel: HDMI set ptbl: 16 2 0 96 7 5 4 1 4 0 0 128 24 88 44 56 45 1 1 
Dec 13 21:42:39 teres kernel: hdmi_clk_disable_prepare()L196
```

I used hdmi through the mini-hdmi to hdmi adapter and then an hdmi to dmi adapter, because I do not have an hdmi monitor.

My config for xorg.conf is basically that of the `rel3` branch, only I added to xorg.conf
```
    Option    "ShadowFB" "false"
```

In the kernel, I made no related changes (I change things like cpu and io governors, f2fs, etc.)

hehopmajieh, 2018-12-17:
Hi Alexei, 
i see this behavior with some adapters, they do not send valid DDC data, i am working on this

closed 2024-06-07 by TsvetanUsunov