OLIMEX/DIY-LAPTOP #31: TERES-PCB5-KEYBOARD: Use keyboard/mouse simultaneously

jcstaudt, 2018-12-10:
Users do not normally have to use the keyboard and mouse simultaneously during everyday computer use. However, this feature is extremely useful when performing tasks such as gaming. In a first-person shooter, for example, it is imperative that you are able to move and look around at the same time.

The TERES-PCB5-KEYBOARD keyboard and mouse devices function one at a time, as if the signals are mux'ed onto the same communication channel. I understand that the TERES-I isn't currently a gaming beast, but it would be highly useful in light gaming applications (such as Minetest, FreeDoom, etc) to use them together with OEM hardware.

An easy work-around is to simply use USB-connected peripherals, but again, it would be nice to use the OEM hardware when these peripherals are unavailable.

DanKoloff, 2018-12-11:
Good suggestions, a little more info about that: this is not a hardware or muxing limitation. This was done on purpose with usability considerations in mind - we found it hard to use the keyboard freely when the touchpad is enabled - the palms were triggering the touchpad accidentally. 

This can be improved by software means in the firmware of the keyboard/touchpad board.

jcstaudt, 2018-12-12:
That's a very good point, and a good decision to do that for better general usability. It sounds like the capacitance cap is not active, where the trackpad goes dead when a threshold is exceeded. I'd have to do some research on the AVR processor to see if this is possible. Additionally, I did notice that there is a dedicated key on the keyboard to enable/disable the trackpad. Maybe this could be used to toggle simultaneous operability instead? However, this modification would then not provide the standard/expected functionality of that key and would be better suited as an optional patch as to not confuse people.

closed 2024-06-07 by TsvetanUsunov