OLIMEX/DIY-LAPTOP #33: Touch-Screen 11.6'' for Pen-Input

rzerres, 2018-12-23:
** Touch-Input Device **
It would be very beneficial if the project could deliver a touch-input panel.
Enrich the portefolio to optionaly substitute the default 11.6'' LCD display, maybe combined with a digital pen or smart pen.

** Handwriting an pen driven notes **
Combining the touch-input panel with an application like "Xournal" would make the TERES a low cost alternative for all students searching for a mobile notebook. 

It would allow them, to take handwritten notes in any lectures and postprocess their notes as needed.
Because of the the small dimensions, its minimal weight and the low power consumtion,  the device would last the day, be handy and applicable for mobile usage.

jcstaudt, 2019-01-01:
I think it is a good idea. Many people consider the Teres a highly-portable notebook great for travel (myself included) and touch-screen capabilities could enhance that use-case. (The next obvious step supporting this concept would be to create a 2-in-1 enclosure, but that would involve researching and pursuing all new injection-molded plastics. Maybe re-purposing an existing platform would be more cost-effective.)

Let's consider the existing plastic enclosure. The current LCD cable runs along the left-hand-side metal hinge. The LCD shown in the link appears to have two ribbon cables more toward the center. Compatibilities/etc aside, modification to the case would be required for proper fitment. However, this could potentially remedy an issue https://github.com/OLIMEX/DIY-LAPTOP/issues/1#issuecomment-377710495. If not using the stock plastic enclosure, it would simply require screw/cabling clearance verification in the 3D-printed design files.

Either way, if this proposal is accepted, there's the process of designing another AV controller board and enabling device compatibility.

rzerres, 2019-01-01:
I dought that we can step forward in progressing the design or enhancement of the given enclosure, if  the project hasn't  defined, how to realize the touch functionality.

- [ ]  add-on  solution
use a given qualtiy proved product to put on top of the actualy used LCD 
(compare this to the given starting point top comment of  #33)

- [ ]  new device
try to find a new product that combines touch and LCD functionality (e.g as used in Microsoft Surface ®)

closed 2024-06-07 by TsvetanUsunov