OLIMEX/DIY-LAPTOP #39: installing os from sd card

Fre-d-die, 2019-06-13:
Hi, I am having booting from a sd card, I have burned khumarahns image of gentoo onto an sd card however when rebooting instead of booting the sd card ubuntu mate starts up as normal, could this be because the sd card is 128 gb? is their a way I can manually choose a boot device?

jcstaudt, 2019-06-28:
When a micro SD card is inserted, the device takes boot priority over onboard storage. However if uboot doesn't locate all expected resources and panics, it'll fall back to onboard storage as a secondary option. Although there are a few variables here, I suspect that the Gentoo image was simply not successfully imaged onto the card.

* What tool/method did you use to image the micro SD card?
* Will the device boot after imaging the card again or using a different card with the same image?
* What model is the 128GB card you used? Does an image file other than the Gentoo image boot properly using that card?

I'm on travel and cannot test khumarahn's image at the moment, but perhaps I can help more directly once I have access to my equipment again.

khumarahn, 2019-06-28:
Hi, @Fre-d-die created two parallel issues. We sorted it out here:
https://github.com/khumarahn/teres1-gentoo/issues/3

closed 2019-06-28 by Fre-d-die