OLIMEX/DIY-LAPTOP #45: Update the design with more efficient CPU

Kreyren, 2022-08-30:
### Is your feature request related to a problem? Please describe.**

Got Pinebook Pro recently and i hate that:
1. I don't have KiCAD files for it to easily add functionality that i need e.g. fingerprint, hotswappable batteries, etc..
2. The RK3399 has spectre and meltdown vulnerabilities and is allegedly using proprietary blobs

### Describe the solution you'd like

My end goal is better, equal or tolerably worse alternative to Pinebook Pro that doesn't have security issues and proprietary blobs.

Can we do better with TERES-1? It's currently using the A64 which is too slow for me

### Describe alternatives you've considered

The IMX8QM-TUKHLA seems to be a good alternative to RK3399 with allegedly no blobs, but it's still not releases and it seems that it's development stalled likely due to blobs..

The next best option seems to be the iMX8QuadPlus which compared to RK3399 has one less Cortex-A72

### Additional context

I have experience with hardware and software development and am willing to help if you can do the brainstorm on what CPU should be used.

TsvetanUsunov, 2022-08-31:
IMX8QM processor alone is EUR 200 this is automotive grade processor and you are not correct, there is co-processor running non free firmware inside it, which takes care of all initializations of memory, etc. NXP provide SDK so you can write your own firmware, but do not have public the sources of the firmware which are used to ship the processors, which may need quite of work. 

For IMX8MQP NXP is working to mainline it, so IMO is more suitable for FOSS OSHW product.

Kreyren, 2023-01-20:
@TsvetanUsunov In hypothetical scenario where we have the documentation needed to know how to make a board for it:

What about SnapDragon Mobile 845?
The PostmarketOS wiki says that it's mainlined[https://wiki.postmarketos.org/index.php?title=Qualcomm_Snapdragon_845_(SDM845)&mobileaction=toggle_view_desktop]

Kreyren, 2023-02-28:
Found RK3588S for which the datasheet is widely available and seems like a good option considering budget and resource efficiency.

In terms of mainlining, the work seems to be ongoing in 6.1

Kreyren, 2023-04-22:
Relevant: https://github.com/starfive-tech/VisionFive2/issues/39

closed 2024-06-07 by TsvetanUsunov

Kreyren, 2024-06-11:
Considered `won't fix`