OLIMEX/DIY-LAPTOP #47: Stage releases for this repository to make 3rd party fabrication easier

Kreyren, 2022-09-15:
This project is pain to fabricate through 3rd party thus proposing changes to address it.

### Expected behavior

Throw the repository at PCB manufacturers/fabricators to get quote for fabrication to choose the best one as fast and painlessly as possible

### Current behavior

When i sent this repository to a PCB Manufacturer then gross majority of them gets confused as it seems that their sales team doesn't know how to use GitHub and KiCAD and ask for additional informations that can be easily fetched from KiCAD.

### Proposed solution
GitHub has "Releases" So stage the following files in there per released version:
* Gerber Data - Ideally with recommended configuration for the fabrication
* Stack-up layout - Where should each layer go as F.Cu .. B.Cu is not easily understandable to some
* NC Drill data
* PCB specifics such as recommended and/or warning for incompatible:
    * materials
        * Thickness of material (for layers)
    * Glass Transition Temperature
    * Drawings
    * Minimal/maximal hole sizes
    * Surface Finish
    * Via process
    * Thickness of the copper traces
    * etc..
* Assembly drawings (for polarity and part orientation)[1][2]

^ These can be automated through Continuous Integration and Delivery management, recommending LaminarCI or WoodpeckerCI

### Additional info

Some manufacturers such as PCBWay do not need assembly drawings, because they have WebUI for the user to specify those which is usually not an option for local manufacturers.

Proposing to make the specifications into a range instead of fixed recommendations to enable optimization of the cost e.g. some manufacturers might prefer different layer heights to provide more economical offer.

### Relevants
1. interactive KiCAD BOM - https://hackaday.com/2018/09/04/interactive-kicad-boms-make-hand-assembly-a-breeze/
2. How to make fabricating drawings in KiCAD 5.0+ https://www.youtube.com/watch?v=9p5sAwXWMno

TsvetanUsunov, 2022-09-15:
why would you send repository with all unnecessary information to PCB producer? They only need drill files and gerbers which are generated from the KiCad with one single mouse click? They layer stackup info in in KiCad DRC rules.
How to use KiCad is essential if one wants to use and contribute.

TsvetanUsunov, 2022-09-15:
> btw. I am happy to passively work on this over time if you like the proposed solution

this would be great, as this is lot of work and I do not see when our developers will have free time to complete

Kreyren, 2022-09-15:
> why would you send repository with all unnecessary information to PCB producer? hey only need drill files and gerbers which are generated from the KiCad with one single mouse click? They layer stackup info in in KiCad DRC rules. -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1247652831)

To ease the 3rd party fabrication and enable people who are not familiar with KiCAD or are unable to use it to be able to get the printed boards to use the notebook.

Because currently many manufacturers don't use KiCAD and as you said only expect Gerbers and Drill files which is not my experience in practice as they need much more as explained in "Proposed solution" which currently makes the board very likely to be mis-fabricated so by providing the releases and indicating that the configuration in those should work for everyone then that makes the project significantly more available.

For example currently i have to pay ~8000 CZK (~327 EUR) for the kit sourcing the parts locally and being able to fabricate the PCBs locally i can get down by around 60% of the cost while also getting more people to get the notebook as fabricating at least 5 PCBs of the mainboard cuts the cost on nearly half of the current 80 EUR (where was this even fabricated it for it to cost this much? it's 8 layer A64 board.. the A64 costs like 4 EUR..).

>  How to use KiCad is essential if one wants to use and contribute. -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1247652831)

I disagree people don't need KiCAD to assemble the notebook and use it and as regular user won't ever have to use it because they are going to get very economical rapairs locally if needs be.. So to me making a set of releases that can be just thrown at the manufacturer by an inexperienced user seems as a good thing.

> this would be great, as this is lot of work and I do not see when our developers will have free time to complete -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1247652831)

Will update the issue with TODO list and will passively address them in my free time then, if you don't mind then i would prefer to use GNU Guile (recommended language for tasks alike by the GNU project) for the CI and using LaminarCI.

The language is based on lisp 1.0 (scheme) so it's generally very easy to learn by semi-experienced developer under few hours (minutes?).

TsvetanUsunov, 2022-09-16:
Although I like your enthusiasm I would first suggest you to manufacture and build one Laptop and then do anything on this repo.
From what I see  you lack basic knowledge for PCBs and manufacturing it's wrong to RP something you still do not understand. We will not merge such PRs.
For instance why change the KiCAD version without anything inside changed? What will happen when KiCAD moves from 6.X to 6.X+1 new PR? What is the purpose new versions of KiCAD always will open the old projects. This is non necessary nor it have sense.
Removing the PCB material FR4 without knowing what is it wrong. 
Trying to translate few words from Bulgarian to English (google will translate pretty decent) but breaking the format and names which is already linked in many documents and pages on our web is also not acceptable.
My advice is: build everything from zero and once you do this successfully do the changes you think are necessary so we can review and merge. Do not guess what is this and that and if it's important or not, if we put it there it's important, if you do  not understand what is it is not ground to remove it :)

closed 2022-09-16 by TsvetanUsunov

Kreyren, 2022-09-16:
Disclaimer: i don't mean to be rude, i want to clarify on the chosen approach.

> Although I like your enthusiasm I would first suggest you to manufacture and build one Laptop and then do anything on this repo.
> From what I see you lack basic knowledge for PCBs and manufacturing it's wrong to RP something you still do not understand. We will not merge such PRs. -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1248950654)

I develop and fabricate PCBs on semi-regulate basis monthly mostly focusing on single, ~4 layers and ~8 layer PCBs and fabrication by hand, but i recognize that i lack practical experience with +10 layers.

> For instance why change the KiCAD version without anything inside changed? -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1248950654)

Because files were last changed 4 years ago and since then kicad changed file format and many things inside it so to me it makes sense to update it now.

> What will happen when KiCAD moves from 6.X to 6.X+1 new PR?  -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1248950654)

In ideal world run tests and merge a new version automatically though CI/CD which is practically impossible to do for KICAD without functional electronic computer assisted development (fECAD) such as the solution used for modeling through OpenSCAD and libFive since the current python scripting backend is immature and not sustainable in production environment.

I am aware of the issue which is why I created https://github.com/OLIMEX/DIY-LAPTOP/pull/49 to standardize the development environment so that 6.0.7 would be used until it's decided that it should be updated again.

FWIW KiCAD people said that they put a lot of effort into making sure that KICAD translates the files from one version to another on version update, but i need to do more research into that until i can in clear mind propose it.

> What is the purpose new versions of KiCAD always will open the old projects. This is non necessary nor it have sense. -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1248950654)

The new version can't open the schematics files without issues and they have to be manually to correctly convert the schematics to pcb designer as they need minor changes for the symbols.

> Removing the PCB material FR4 without knowing what is it wrong. -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1248950654)

I removed it from BOM, because from my experience it's no longer being included in it as it's cost varies per manufacturer which seems to be the same reasoning to why KiCAD is not including it in the generated BOM.. In general manufacturers just want to know the list of components that they will put on the board.

>  Trying to translate few words from Bulgarian to English (google will translate pretty decent) but breaking the format and names which is already linked in many documents and pages on our web is also not acceptable. -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1248950654)

Open-source the website so that the changes can be made with the merge request then? Or adjust the wording so that it refers to the file considering that it now has a documentation and technical drawing?

> My advice is: build everything from zero and once you do this successfully do the changes you think are necessary so we can review and merge. Do not guess what is this and that and if it's important or not, if we put it there it's important, if you do not understand what is it is not ground to remove it :) -- @TsvetanUsunov (https://github.com/OLIMEX/DIY-LAPTOP/issues/47#issuecomment-1248950654)

I don't touch things i don't understand.. if that happened i would highlight the area in schematics/pcb designer and asked for clarification, what i did is added technical drawings and documentation and then launched two kicads and compared layer to layer to make sure the version update didn't corrupt anything.

Btw. Did you miss that i closed https://github.com/OLIMEX/DIY-LAPTOP/pull/48 (that had corruption due to the schematic issues) and replaced it with https://github.com/OLIMEX/DIY-LAPTOP/pull/50 ?

Kreyren, 2022-09-16:
Talking about the issue with kicad version management with kicad people the proposed approach was to make a CI/CD implementation that generates gerbers and schematics from two different versions and then runs diffoscope on them to capture any irregularities across versions which i was told is sustainable in production, but would need to research it more.

Was told that version should be updated every 2 weeks ish which is +- what i do in my development as well.