OLIMEX/DIY-LAPTOP #51: Troubleshooting link is broken in README

xloem, 2022-09-20:
Just a note regarding this repository.

xloem, 2022-09-20:
https://github.com/OLIMEX/DIY-LAPTOP#other-resources
https://github.com/OLIMEX/DIY-LAPTOP/tree/master#other-resources

jcstaudt, 2022-09-20:
Fixed on the `rel3` branch in commit 6f9910d.

closed 2022-09-20 by jcstaudt

xloem, 2022-09-21:
Thanks. It looks like the original plan was to have documentation for diagnosing issues. It’s too bad we didn’t all make this happen, me included.

jcstaudt, 2022-09-21:
Agreed. Yes, you're right; my initial plan was to browse the most common or annoying hurdles seen in the forums and attempt to aggregate them into some organized structure which would resemble an operator's manual. I'm a bit limited in my free time lately to do this entirely on my own, but I'm happy to assist if you're willing to help or if there's an expressed need from the community.