OLIMEX/DIY-LAPTOP #52: Expand the BOOT partition size on the official image to fix upgrades

Kreyren, 2023-01-20:
[The official image for teres-1](https://ftp.olimex.com/Allwinner_Images/A64-Teres/linux/xenial-teres-bspkernel-20190121_144941_EET-1_hdmi_out_v1_5.zip) is unable to successfully upgrade distribution release, because the BOOT partition has 50MB where invoking `$ do-release-upgrade` requires additional ~14MB and doing GUI `update-manager` will demand additional ~48.14MB and upgrading from bionic seems to need around 150MB.

### Steps to Reproduce
1. Boot the image
2. Do:
* `# apt update && apt upgrade -y && apt dist-upgrade -y && reboot`
* `do-release-upgrade`

### Proposal
Expand the boot partition from 50MB -> 200MB

---

CC @TsvetanUsunov

DanKoloff, 2023-01-20:
https://github.com/OLIMEX/DIY-LAPTOP/blob/master/doc/web/sw_updating-os.md

"Important! Currently the image upgrade is possible only via a clean install (via SD card). Running "sudo apt upgrade" would lead to unexpected results and problems! Avoid running "apt upgrade"!"

Kreyren, 2023-01-20:
> master/doc/web/sw_updating-os.md -- @DanKoloff (https://github.com/OLIMEX/DIY-LAPTOP/issues/52#issuecomment-1398246623)

I understood from that i am supposed to install that OS on the emmc and then do the upgrade from there no?

Btw. The steps on https://github.com/OLIMEX/DIY-LAPTOP/blob/master/doc/web/sw_fresh-os.md using `install_emmc.sh` failed for me as well so i downloaded the image on the OS booted from the SDcard and then flashed it on internal storage using `dd`

DanKoloff, 2023-01-20:
> I understood from that i am supposed to install that OS on the emmc and then do the upgrade from there no?

Seriously, how ambiguous can "Avoid running "apt upgrade"!" be?

Kreyren, 2023-01-21:
@DanKoloff I was talking about upgrading the OS as described in https://github.com/OLIMEX/DIY-LAPTOP/blob/master/doc/web/sw_fresh-os.md

1. The `install_emmc.sh` doesn't work for me reliably
2. Doing `dd` from booted SDCard to internal storage results in corrupted image

So the only way that works for me now is to just use the SDCard and upgrade the OS which requires ~300MB of boot partition to succeed (up until jammy that breaks the system)

In relation to https://github.com/OLIMEX/DIY-LAPTOP/blob/master/doc/web/sw_fresh-os.md?rgh-link-date=2023-01-20T12%3A12%3A08Z the image has very outdated kernel, OS and non-functional web browser along side majority of packages so `apt update` doesn't really cut it

Kreyren, 2024-06-11:
Considered `won't fix`