OLIMEX/DIY-LAPTOP #55: Proposals for the next design

Kreyren, 2023-03-01:
Context: I am working on a custom magnesium case for the teres's next chip and i want to ideally just adjust the daughter boards instead of having to maintain fork of the mainboard + making it less painful to add features such as RFID, hot-swappable batteries etc..

---

The idea of these proposals is to make the current design easily expandable and changable while reducing the need to rework PCBs for e.g. use of different case and devices.

### Standalone right-side I/O

Currently the mainboard and right side I/O are the same board:

[image]

Instead proposing to split the right side I/O as it's case-specific feature that limits modularity.

[image]

### Adjust SBC designs

Adjust the SBC designs so that they can be a drop-in replacement in teres without affecting the functionality of the SBC itself.

in real world example: deprecate the TERES-PCB1-A64 and rework OlinuXino-A64 (not suggesting to do that now, but for future chips)

### Add edge connector for UEXT

Currently to add devices, one has to scrape solder mask and solder wires to the mainboard's bus so instead proposing to add UEXT connector so that devices can be easily added in.

The idea is that device's PCBs can then be connected to UEXT and ancored inside the case with e.g. hot glue, double-sided tape or adjusting the case design.

---

So overall the design should imho be just a chipless case that people can choose which chip they want to have in it.

---

CC @TsvetanUsunov

TsvetanUsunov, 2024-06-07:
I already think for re-design but not in the same plastic case, stay tuned it will pop on our blog :)

closed 2024-06-07 by TsvetanUsunov

Kreyren, 2024-06-11:
> I already think for re-design but not in the same plastic case, stay tuned it will pop on our blog :) -- @TsvetanUsunov

Can you please consider making it some functional SOM so that we can have 2~4 layer platform PCBs that can be easily fabricated by the users including locally with the SOM  being less than 50x50mm (bcs 3rd party will make 8 layer PCBs that are very economical e.g. 2 EUR for 5 that way) which can just have the CPU RAM and ideally serial console so that it can work independently to slot into the body to easily change the chips? It would make the design a lot more usable.