OLIMEX/DIY-LAPTOP #58: Add an ability to poweron/wake up the system

Kreyren, 2023-03-06:
Currently i always have to get up and turn the system on or open the lid when i want to access it remotely -> Proposing to implement some way to power it on/wake up to manage this scenario

haraldg, 2023-04-10:
Isn't that something that needs to be handled in the driver for RTL8723BS?

Kreyren, 2023-07-16:
> Isn't that something that needs to be handled in the driver for RTL8723BS? -- @haraldg

Apparently so but i wasn't able to get it to work so far thus why am i annoying tsvetan :D

haraldg, 2023-07-16:
I think there are basically 3 possible reasons why it doesn't work:
1. The RTL8723BS driver doesn't actually support this yet.
2. The driver would work, but the chip gets powered down too far and can not properly process it.
3. The WL-WAKE-AP E13 pin actually gets triggered but the system fails to wake up.

(1) is clearly a driver problem and can only be addressed there. (2) might be a problem with the HW design or the devicetree file. (3) might be a problem with the devicetree file or some sunxi specific code. It might also need some support from https://github.com/crust-firmware/crust to get this to work.

Obviously the first step forward is to figure out, which of the above it is.

closed 2024-06-07 by TsvetanUsunov