OLIMEX/DIY-LAPTOP #61: Add resetable fuse for 5VDC3A on the Vin

Kreyren, 2023-04-01:
Referencing https://www.olimex.com/forum/index.php?topic=7782.0

Proposing to integrate **resetable fuse** for 5VDC3A on the Vin

Options: https://eu.mouser.com/c/circuit-protection/thermistors/resettable-fuses-pptc/?current%20rating%20-%20max=3%20A&maximum%20voltage=5%20V

So that if the user overvoltages the laptop it won't damage anything

DanKoloff, 2023-04-03:
I don't we would place fuse by default. We have provided the option to add fuse, user can cut between the pads of FUSE2 and add such there. Refer to the picture below

[image]

closed 2023-04-03 by DanKoloff

Kreyren, 2023-04-03:
@DanKoloff i think you missunderstod, the proposed fuse (e.g. https://eu.mouser.com/ProductDetail/onsemi/NIV6350MT2TXG?qs=xZ%2FP%252Ba9zWqa2xJAD3VNXxQ%3D%3D) seems to be a 0.659 EUR integrated circuit component that automatically cuts power when it goes past the 5VDC3A otherwise it lets power go through, the reset it handled by the fuse itself without the need of user doing anything.

FUSE2 is not an option without developing a daughter board on top of it as the pining is not compatible:

[image]

Kreyren, 2023-09-22:
@DanKoloff Did you read the reply?

xloem, 2024-06-07:
Noting it sounds like olimex does not want to add a fuse, but aren’t there 2-lead resettable fuses? What is the advantage of an entire integrated chip?

It would be super nice to have a fuse there by default, prevent board damage.

TsvetanUsunov, 2024-06-07:
I do wonder why additional fuse is necessary if the power is supplied with Adapter which already have all the protections?

xloem, 2024-06-07:
Some of us prefer to power the laptop with DC. https://linux-sunxi.org/Olimex_Teres-A64#Cut_the_power_cable
I saw the same topic somewhere else on cutting the power cable, not certain where.

khumarahn, 2024-06-07:
I cannot use the stock power adapter for teres: it only fits the European power socket, it is bulky, and I travel quite a lot. I use instead a usb cable and charge from various usb chargers.

I never had any problems with this, but I don't think that it is safe to rely on protections of the power adapter.

Kreyren, 2024-06-11:
> Some of us prefer to power the laptop with DC. https://linux-sunxi.org/Olimex_Teres-A64#Cut_the_power_cable I saw the same topic somewhere else on cutting the power cable, not certain where. -- @xloem (https://github.com/OLIMEX/DIY-LAPTOP/issues/61#issuecomment-2154824652)

It's me who wrote that there as i basically charge the laptop from any kind of power source that i can find.. e.g. i often splice the cables together with WAGO connectors to charge in the car, friend's phone, escooter or even stuff like disassmbling street light and connecting aligator clips to the power there to charge it (.. when really desperate and the situation needs it..)

Afaik other teres users do the same just usually not as extreme e.g. charging from a portable solar panel is apparently very common which afaik has an increased risk for spikes in voltage/current.

also as said in https://github.com/OLIMEX/DIY-LAPTOP/issues/67 the stock adapter is useless shit.

FWIW also currently working on a standard to get a screwable connectors e.g.  YC8 (https://www.youtube.com/watch?v=stdiyrQk3VU) to deprecate the need for WAGOs in my teres development, but i am struggling to figure out how to make them water-proof and non-annoying to use.

Kreyren, 2024-06-11:
Additionally: Not completed and moved in https://git.dotya.ml/KREYLIMEX/TERES/issues/23