OLIMEX/DIY-LAPTOP #66: Provide the firmware for RTL8723BS in the repository

Kreyren, 2023-09-18:
The current design is dependent on that file to get the WiFi and BLE to work

DanKoloff, 2023-10-03:
I don't think we have it. I think it has Realtek firmware and the firmware sources probably are not open source. Drivers should be available tho.

Kreyren, 2023-10-03:
@DanKoloff There seems to be sources for the driver, but i wasn't able to figure out if it's open-source or not e.g. https://github.com/anthonywong/rtl8723bs

Additionally it's in [linux-firmware](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git) so at very least i would at least provide a text file pointing out where to find it.

Kreyren, 2023-10-03:
Actually it seems to be there in https://github.com/torvalds/linux/tree/cbf3a2cb156a2c911d8f38d8247814b4c07f49a2/drivers/staging/rtl8723bs and under GPLv2.. i try to mess with it..

Kreyren, 2023-10-03:
oh i seem to already have this running on my experimental NixOS build so yes it seems that the driver if open-source and mainlined

closed 2023-10-03 by Kreyren