OLIMEX/DIY-LAPTOP #67: Won't charge anymore after extensive daily use

Kreyren, 2023-09-22:
I've been using my teres every day since i got it and now i have to angle the charging port in weird ways for it to have contact and charge..

The issue is only on the male side so i just cut the cable and spliced a new connector with WAGO, but it's still pretty annoying..

Proposing for the next version to use more robust one

DanKoloff, 2023-09-25:
We also replace these connectors on the test computers from time to time. They indeed start to expire after around 1000 times plugging and unplugging.

Kreyren, 2023-09-26:
> We also replace these connectors on the test computers from time to time. They indeed start to expire after around 1000 times plugging and unplugging. -- @DanKoloff (https://github.com/OLIMEX/DIY-LAPTOP/issues/67#issuecomment-1733634230)

So USB-C next? Or is there any more rugged connector you can recommend? (making a new PCB for teres so brainstorm would help)

xloem, 2024-06-01:
I use SAE 12v and IDE 12/5v.

I think with regard to ruggedness the key is to figure out how to get stress to load the casing rather than solder joints or contacts.

closed 2024-06-07 by TsvetanUsunov

Kreyren, 2024-06-11:
Considered `won't fix`