OLIMEX/DIY-LAPTOP #68: Fastboot?

Kreyren, 2023-10-06:
Reading through the docs it seems that the A64 should be capable of Fastboot mode, but there doesn't seem to be any way to enter this mode on teres?

The fastboot is expected to be used to flash things directly in the eMMC.

CC @DanKoloff

DanKoloff, 2023-10-09:
Not sure if Linux has good support for fastboot at all. I think it is some Android thing that maybe isn't well supported in Linux at all.

Kreyren, 2023-10-10:
> Not sure if Linux has good support for fastboot at all. I think it is some Android thing that maybe isn't well supported in Linux at all. -- @DanKoloff (https://github.com/OLIMEX/DIY-LAPTOP/issues/68)

it's a communication protocol not some android thing.. android ppl just use it often..

Linux is it's bootloader level thing, see https://gitlab.com/pine64-android/brandy/blob/master/u-boot-2014.07/usb_sunxi/usb_fastboot.c and https://www.mail-archive.com/linux-sunxi@googlegroups.com/msg18696.html and uboot implementation https://github.com/u-boot/u-boot/blob/d9bb6d779b69c2548891e568e5e2a23e1b7eedaa/cmd/fastboot.c#L60

also https://linux-sunxi.org/Fastboot

The general practice is that there is a physical button on the board that is held down (teres has a convinient hole at the back of the chasis that is not used) and then connecting an USB to get in fastboot mode or `fastboot usb 0` from uboot cmdline (assuming that it's compiled in)

Alternatively there is also https://linux-sunxi.org/U-Boot/USB_Mass_Storage to get write access to the eMMC in theory

Kreyren, 2023-10-11:
Did some research in this and thanks to #linux-sunxi:

A64 has 3 known ways to get write access to the emmc:
1. Proprietary FEL protocol that is used as a last resort in case the device is hard bricked and should be avoided otherwise, it can be accessed through the FEI pin on the chip which is non-interrupt
2. Fastboot which can be used by compiling that in u-boot with an option `CONFIG_USB_MUSB_GADGET=y`
3. UMS which can be also compiled in u-boot with at least `CMD_USB_MASS_STORAGE=y`

the sunxi wiki also mentions USB 2.0 OTG port that teres doesn't seem to have implemented in it's DTS with a declaration alike:

```
&usb_otg {
	dr_mode = "host";
	status = "okay";
};
```

Additionally the daughter board that this port is routed into should probably have a thicker trace for the USB5V and be labeled correctly as USB5VOTG

closed 2024-06-07 by TsvetanUsunov