OLIMEX/DIY-LAPTOP #73: FEL doesn't work

Kreyren, 2024-05-18:
It's not possible to enter [FEL](https://linux-sunxi.org/FEL) through holding **UBOOT** button and the powering up with the USB cable connected to the USB-OTG.

Kreyren, 2024-05-19:
Seems that it doesn't even let me in FEL when requested from uboot:

```
-Boot SPL 2024.01-armbian (Feb 09 2024 - 18:59:43 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.9(debug):armbian
NOTICE:  BL31: Built : 18:58:03, Feb  9 2024
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x20a4700, model: Olimex A64 Teres-I
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP803 on RSB
INFO:    PMIC: aldo1 voltage: 2.800V
INFO:    PMIC: dcdc1 voltage: 3.300V
INFO:    PMIC: dcdc5 voltage: 1.500V
INFO:    PMIC: dcdc6 voltage: 1.100V
INFO:    PMIC: dldo1 voltage: 3.300V
INFO:    PMIC: dldo2 voltage: 2.500V
INFO:    PMIC: dldo3 voltage: 1.200V
INFO:    PMIC: dldo4 voltage: 3.300V
INFO:    PMIC: fldo1 voltage: 1.200V
INFO:    PMIC: Enabling DC SW
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is available via SCPI
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
ns16550_serial serial@1c28000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

U-Boot 2024.01-armbian (Feb 09 2024 - 18:59:43 +0000) Allwinner Technology

CPU:   Allwinner A64 (SUN50I)
Model: Olimex A64 Teres-I
DRAM:  2 GiB
Core:  74 devices, 24 uclasses, devicetree: separate
WDT:   Not starting watchdog@1c20ca0
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Net:   No ethernet found.
230454 bytes read in 12 ms (18.3 MiB/s)
starting USB...
Bus usb@1c1b000: USB EHCI 1.00
scanning bus usb@1c1b000 for devices... 4 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Autoboot in 1 seconds, press <Space> to stop
=> go 0x10000
## Starting application at 0x00010000 ...
"Synchronous Abort" handler, esr 0x02000000
elr: ffffffff8c0bf004 lr : 000000004a00dbb8 (reloc)
elr: 0000000000010004 lr : 00000000bdf5ebb8
x0 : 0000000000000001 x1 : 00000000b9f41238
x2 : 00000000b9f41238 x3 : 0000000001c28000
x4 : 0000000000000000 x5 : 0000000000000004
x6 : 00000000bdfe0e50 x7 : 0000000000001558
x8 : 0000000000000010 x9 : 0000000000000008
x10: 0000000000000001 x11: 0000000000000001
x12: 0000000000000010 x13: 00000000be1ff3c8
x14: 0000000000000008 x15: 0000000000000000
x16: 0000000000010000 x17: 0000000000000004
x18: 00000000b9f30d90 x19: 00000000b9f41238
x20: 0000000000010000 x21: 0000000000000002
x22: 0000000000000001 x23: 0000000000000002
x24: 00000000bdff60c4 x25: 0000000000000000
x26: 0000000000000000 x27: 0000000000000000
x28: 00000000b9f40560 x29: 00000000b9f26600

Code: 00000000 00000000 00000000 ea000016 (4e4f4765)
Resetting CPU ...

resetting ...

U-Boot SPL 2024.01-armbian (Feb 09 2024 - 18:59:43 +0000)
...
```

DanKoloff, 2024-05-29:
I believe you can create SD card that forces FEL. You need to check the A64 article about FEL:

https://linux-sunxi.org/A64#FEL_booting

Kreyren, 2024-05-29:
> I believe you can create SD card that forces FEL. You need to check the A64 article about FEL: 
>
> [linux-sunxi.org/A64#FEL_booting](https://linux-sunxi.org/A64#FEL_booting) -- @DanKoloff (https://github.com/OLIMEX/DIY-LAPTOP/issues/73#issuecomment-2137017844)

I didn't have success with the SD Card, notice the:

```
=> go 0x10000
## Starting application at 0x00010000 ...
"Synchronous Abort" handler, esr 0x02000000
elr: ffffffff8c0bf004 lr : 000000004a00dbb8 (reloc)
elr: 0000000000010004 lr : 00000000bdf5ebb8
x0 : 0000000000000001 x1 : 00000000b9f41238
x2 : 00000000b9f41238 x3 : 0000000001c28000
x4 : 0000000000000000 x5 : 0000000000000004
x6 : 00000000bdfe0e50 x7 : 0000000000001558
x8 : 0000000000000010 x9 : 0000000000000008
x10: 0000000000000001 x11: 0000000000000001
x12: 0000000000000010 x13: 00000000be1ff3c8
x14: 0000000000000008 x15: 0000000000000000
x16: 0000000000010000 x17: 0000000000000004
x18: 00000000b9f30d90 x19: 00000000b9f41238
x20: 0000000000010000 x21: 0000000000000002
x22: 0000000000000001 x23: 0000000000000002
x24: 00000000bdff60c4 x25: 0000000000000000
x26: 0000000000000000 x27: 0000000000000000
x28: 00000000b9f40560 x29: 00000000b9f26600

Code: 00000000 00000000 00000000 ea000016 (4e4f4765)
Resetting CPU ...

resetting ...

U-Boot SPL 2024.01-armbian (Feb 09 2024 - 18:59:43 +0000)
```

Which afaik does the same thing.. Even when it's forced it seems to cause the chip to crash
-> Did OLIMEX do any modification to the invidual chips? It's reported to work on other A64s 
-> Can you look into the docs you have from allwinner to see what might be causing this as i couldn't find a rationale in the publicly available once and it doesn't seem to be something trivial like current leakage in the reset circuit.

DanKoloff, 2024-05-30:
We haven't modified any chips. We don't have anything extra, all we've got we've published online. FEL is inside the chip and it can't be affected by hardware or software. 

How did you prepare that card? First make a the simplest test with only the fel-sdboot.sunxi on the card. It is not meant to be a conclusive solution, just to test entering FEL via SD card. Download this:

https://github.com/linux-sunxi/sunxi-tools/blob/master/bin/fel-sdboot.sunxi

Then download it to your card with:

dd if=fel-sdboot.sunxi of=/dev/xxxxxx bs=1k seek=8

where /dev/xxxxxx is the location of the SD card as assigned by Linux.

Also maybe check here https://linux-sunxi.org/FEL/USBBoot

closed 2024-06-07 by TsvetanUsunov