OLIMEX/DIY-LAPTOP #75: [Question] What is the part that is used as the touchpad ?

tranquillity-codes, 2025-01-25:
The site just states [TERES-023-Touch](https://www.olimex.com/Products/DIY-Laptop/SPARE-PARTS/TERES-023-Touch/), and we can't find any info about what part it is, as well as any kind of traces and information about the part - all we can find are 3 pictures, 1 in this repository (images/TERES-I/hardware/043.jpg) and two low-res ones on the site.

Kreyren, 2025-01-25:
Afaik it's generic off the shelf touchpad that came with the chasis that is then routed to the chip via https://github.com/OLIMEX/DIY-LAPTOP/tree/rel3/HARDWARE/A64-TERES/TERES-PCB3-TOUCH-BTN/Rev.A (verified on my teres)

@TsvetanUsunov or @DanKoloff probably have a reference somewhere what exact part it is (I kinda put it in place with a very strong tape and now i can't remove that non-destructively without a lot of heat and solvents to check as the original glue seemed kinda too weak to me :D) , but if you are looking for a reference or just want to make your own then https://ploopyco.github.io/trackpad/assembly-instructions/ should have everything you need and you can adjust it's dimensions to fit in the teres's chasis.

tranquillity-codes, 2025-01-25:
The chassis is custom-made, no ? Or is it off-the-shelf ? Would be nice to figure out what that part is too, I suppose, if it's off-the-shelf...

Would also be interesting to know where is OLIMEX sourcing it from

DanKoloff, 2025-01-27:
There is a repository with reverse engineering of the plastic case made by another company, maybe it would be useful:

[Teres case reverse engineered](https://code.forksand.com/forksand/olimex-teres-case)

We don't have source files for the plastic case and such. We purchased the TERES-I case ourselves but we never got design files from the manufacturer. So we first got the case then designed the insides to fit in this case and we empirically measured and tested.

Kreyren, 2025-01-27:
> The chassis is custom-made, no ? Or is it off-the-shelf ? Would be nice to figure out what that part is too, I suppose, if it's off-the-shelf... -- @tranquillity-codes (https://github.com/OLIMEX/DIY-LAPTOP/issues/75#issuecomment-2614002018)

The chasis is off the shelf with there being a community alternatives that are open-source and very economical to get them made as the teres's size makes it to not fit in common low-end 3D printers and afaik most of the teres users seems to design their own chasis as the design is simple and can be made in few mins.